DT42/BerryNet #2: Failed to start detection_server.service

bafu, 2017-05-13:
Reported by L kouchla <kouchlah@gmail.com>:

Hi;

I executed the command ./configure and the instllation is completed. But I think some packages or files are missing during the installation.

When I run "berrynet-manager start", it tells me that it does not find the commnade.

When I run "./berrynet-manager start". it says :

Failed to start detection_server.service: Unit detection_server.service failed to load: No such filectory.

Services: broker, dashborad, localim, camera, newspaper and cleaner failed to start.

What am I doing wrong ?

Is ther a step by step tutorial for installation

Thank you

samrudh, 2017-05-13:
I also got same error. But I am using Pi2. Is it because of that?
Which raspberry pi version is supported?

bafu, 2017-05-13:
@samrudh Sorry for hitting this problem. I'm trying to reproduce it. If this issue exists, it should be HW-independent.

BerryNet is only tested on RPi3 currently.

arvdsar, 2017-05-13:
I have the same issue on RPi3. Clean OS install (2017-04-10-raspbian-jessie-lite), apt-get update and upgrade. Next I've ran the setup. 
Failed to start all services with message: No such file or directory

bradleyscott, 2017-05-13:
I also have the same issue. I did some debugging but didn't get quite to the bottom of it. It looks like its file path / file reference problems. 
FIrst of all, the .system files weren't in /etc/systemd/system and I tried manually moving there, and afterwards berrynet-manager started working.
However, when running berrynet-manager status each service seems to be searching for the .js files associated with them. For example, camera.service is looking for camera.js but can't resolve the file when its spinning up. I am not clear if it expects to to be in /etc/systemd/system or /usr/bin/node but either way it just doesn't seem like everything is referencing where it should after the ./configure process finishes.

arvdsar, 2017-05-13:
Seems you also need to:
sudo cp -a broker.js camera.js cleaner.sh config.js dashboard inference journal.js localimg.js mail.js package.json /usr/local/rpi3-ai-gateway

The files are located in ~/Berrynet and were not copied.
Next go to /usr/local/rpi3-ai-gateway and run:
sudo npm install 
Now a bunch of files are build in /usr/local/rpi3-ai-gateway/node_modules

Seems the configuration script fails from the line: install_systemd_configs()  and further

Not sure if all is done now, but at least a little progress ;-)

bafu, 2017-05-13:
The root cause is that the systemd service files are installed in the correct directory. The installer enters a wrong directory and can not find them.

closed 2017-05-13 by bafu

bafu, 2017-05-13:
@arvdsar @bradleyscott Sorry for hitting the problem. Here are more details: installer (configure) stayed in a wrong directory after the `download_detector_model` function.

The installation steps of dashboard, systemd configs, and gateway files (.js) are after this function, so all of their installation are failed (bad!). Thank you for your feedback!

commit 97c88b7f (2017-05-13): Fix the installation issue (close #2).