DT42/BerryNet #3: IP camera access to user name and password

ytgcljj, 2017-05-19:
IP camera access to user name and password，I don't know how to set it up

bafu, 2017-05-20:
@ytgcljj The way to feed username and password info depends on your camera. If your camera is in the [camera connection database](https://www.ispyconnect.com/sources.aspx), you can try the generated link.

If your camera is not in the database, you can check the camera's user manual, or contact the vendor for the access details.

bafu, 2017-05-21:
If this issue still exists, please feel free to reopen this issue.

closed 2017-05-21 by bafu

ytgcljj, 2017-05-22:
thank you ,let me try

kouchla, 2017-05-22:
@bafu: In which file should I configure my IP cam (config,js or camera.js or other file) I kowon how  to feed username and password with my camera but I don,t know in which BerryNnet's file

Thank's with you help

ytgcljj, 2017-05-23:
@kouchla  can you tell me how to set up the ip camera with username and password? 
  i visited the camera ip http://192.168.254.4/doc/page/login.asp?_1495505377878

kouchla, 2017-05-23:
@ytgcljj  With my VLC player, I can see the flow of the camera (DLink) by doing this:
http://username:password@ipcanera address:port/video.cgi

With Foscam cameras, replace video.cgi with videostream.cgi (I think).

I hope it helps.