DT42/BerryNet #4: Add Video Analysis

hskang9, 2017-05-21:
is Berrynet able to detect objects in video or while streaming? Even though fps may be lower because of the hardware, it would work on other IoT device specialized in computer vision(Nvidia Jetson, Intel Joule, etc).

bafu, 2017-05-21:
@hskang9 Yes, this feature is in the project roadmap. I totally agree with you that this feature will be useful if BerryNet runs on a more powerful HW platform, or the real-time demand is not very strong in a scenario.

YingHou, 2017-09-14:
Hi Bafu, I was wondering what's the timeline to add this feature in?

bafu, 2017-10-09:
@hskang9 @YingHou Video streaming support is available in 2.1.0. Please try the new feature and any suggestions is welcome, thanks!

closed 2017-10-09 by bafu