DT42/BerryNet #8: results of ipcamera

decombas, 2017-09-26:
I have installed berrynet and it works well with an image. But when I try with my IP camera ( heden 2.2) it doesn't work.
I have modified the config like that :
ifconfig eth0 192.168.0.1 netmask 255.255.255.0

then check that i work on firefox :
http://192.168.0.2:8080/videostream.cgi?user=<USER>&pwd=<PWD>
http://192.168.0.2:8080/snapshot.cgi?user=<USER>&pwd=<PWD>

I modified the config.js file  ( config.ipcamerashapshot )

I then start berrynet-manager start
go to : http://localhost:8080/index.html#source=dashboard.json
and start the camera :
mosquitto_pub -h localhost -t berrynet/event/camera -m snapshot_ipcam

in MQTT client logs, I have : camera client : on topic/event/camera, received message snapshot_ipcam.
but no visual result.

Can someone help me to know if the problem come from my camera ? the configuration of the dashboard ? something else ?
Is there a folder where the images are saved ?

Thank you

GitHubForNH2, 2017-10-30:
excuse me, I have installed berrynet too, but i have not understood how to let it work with  an image, could you held me?(i have run "./configure" and "mosquitto_pub -h localhost -t berrynet/event/localImage -m ./dog.jpg" but i can't get anything but MQTT CLIENT LOG,where can i find the result?)

bafu, 2017-11-04:
@decombas Thank you for interesting in BerryNet and sorry for replying late. Can you share `/var/log/syslog` with me and I will check what happened in your system.

@GitHubForNH2 I replied the question on issue #9 and please let me know if you still can not resolve the issue, thanks.

bafu, 2019-10-27:
BerryNet keeps improving and fixing issues. If this issue still exists in the latest release, please share the steps to reproduce with me, and I will investigate it.

Please feel free to re-open this issue if it still can be reproduced.

closed 2019-10-27 by bafu