DT42/BerryNet #9: what do i need to do when run ./configure?

GitHubForNH2, 2017-10-30:
i don't konw how to get a result from this project. when i run "./configure" i haven't get anything but MQTT CLIENT LOGS , i don't konw where can i get a result when i run "mosquitto_pub -h localhost -t berrynet/event/localImage -m ./dog.jpg"

bafu, 2017-11-04:
@GitHubForNH2 You can see the inference result on the dashboard. So here are the high-level steps:

```
# Step 1: Install
rpi3 $ ./configure

# Step 2: Launch BerryNet
rpi3 $ berrynet-manager start

# Step 3: Launch dashboard, you can use your notebook's browser

# Step 4: Execute MQTT publish command to send an image as input
rpi3 $ mosquitto_pub -h localhost -t berrynet/event/localImage -m ./dog.jpg

# Step 5: See the inference result on dashboard
```

Please let me know if you still have any issue, thanks.

muzi2045, 2018-01-13:
when I run this:
   `$ mosquitto_pub -h localhost -t berrynet/event/localImage -m ./dog.jpg`
I got a Error: Connection refused
I don't know what happended.
Is there anyone explain it ?

yimingliu0216, 2018-01-17:
I am confusing here too. Does the project has mqtt broker build in? Or it needs a mqtt broker on the pi?

dhrubins, 2018-01-20:
@muzi2045, I had this same problem. I installed MongoDB, set it up to start on boot, and everything started to work.

muzi2045, 2018-02-18:
@dhrubins, thanks for your advice , I'll try it .

bafu, 2018-07-09:
@GitHubForNH2 @muzi2045 @yimingliu0216 @dhrubins The [v2.3.0 installation image](https://github.com/DT42/BerryNet/releases/tag/v2.3.0) is available now, and it will save community friends' efforts to build from scratch. Please try it and feel free to let me know if there is any issues, thanks!

bafu, 2018-07-09:
@yimingliu0216 BerryNet project is using [Mosca](https://github.com/mcollina/mosca) as default broker, and we are surveying the performance and compatibility of [EMQ](http://emqtt.io/).

closed 2018-07-09 by bafu