DT42/BerryNet #10: Simply Doesnt Work !!

gmonk, 2017-11-01:
Nothing actually works :-1:   Followed directions and the dashboard doesnt show anything.  looking at the directory I can see there are serveral snapshots but thery are all the same image with different time stamps

bafu, 2017-11-04:
@gmonk Thanks for the issue report. Can you share your `/var/log/syslog` with me and I will check this issue? Thanks.

GitHubForNH2, 2017-12-02:
i also meet the same problem and this is my syslog file:
[syslog.zip](https://github.com/DT42/BerryNet/files/1523839/syslog.zip)

bafu, 2017-12-14:
@GitHubForNH2 In your syslog, I found the error message `Error: listen EADDRINUSE :::1883`, which means that BerryNet MQTT broker's default port has been registered by another service, e.g. mosquitto MQTT server.

In this case, there are two ways to resolve this issue:

1. Disable mosquitto MQTT server by `sudo service mosquitto stop`.
1. Change BerryNet MQTT broker's default port in broker.js.

and then restart BerryNet.

I will close this issue and please free to re-open it if this issue can not be resolved, thanks.

closed 2017-12-14 by bafu