DT42/BerryNet #11: detectord.py OSError: [Errno 2] No such file or directory

datajerk, 2017-11-15:
```
Nov 15 02:51:36 raspberrypi bash[18053]: SNAPSHOT_DIR: /usr/local/berrynet/inference/image
Nov 15 02:51:36 raspberrypi bash[18053]: [1510714296.905] /tmp/detectord.pid exists but process 18048 died, clean it up.
Nov 15 02:51:36 raspberrypi bash[18053]: [1510714296.906] watch_dir:  /usr/local/berrynet/inference/image
Nov 15 02:51:36 raspberrypi bash[18053]: [1510714296.906] pid:  /tmp/detectord.pid
Nov 15 02:51:36 raspberrypi bash[18053]: Traceback (most recent call last):
Nov 15 02:51:36 raspberrypi bash[18053]: File "detectord.py", line 137, in <module>
Nov 15 02:51:36 raspberrypi bash[18053]: stderr=subprocess.STDOUT)
Nov 15 02:51:36 raspberrypi bash[18053]: File "/usr/lib/python2.7/subprocess.py", line 710, in __init__
Nov 15 02:51:36 raspberrypi bash[18053]: errread, errwrite)
Nov 15 02:51:36 raspberrypi bash[18053]: File "/usr/lib/python2.7/subprocess.py", line 1335, in _execute_child
Nov 15 02:51:36 raspberrypi bash[18053]: raise child_exception
Nov 15 02:51:36 raspberrypi bash[18053]: OSError: [Errno 2] No such file or directory
```

Where's `./darknet`?

datajerk, 2017-11-15:
```
Installation is completed successfully!
```

Yet:

```
compilation terminated due to -Wfatal-errors.
Makefile:99: recipe for target 'obj/image.o' failed
make: *** [obj/image.o] Error 1
make: *** Waiting for unfinished jobs....
```

Darknet patches also failed.

Perhaps some error detection is in order.

bafu, 2017-11-17:
@datajerk Thank you for trying BerryNet and reporting the issue report. I will check it and share my investigation result with you soon.

bafu, 2017-11-30:
This issue should be fixed in [v2.1.1](https://github.com/DT42/BerryNet/releases/tag/2.1.1). Please feel free to reopen this issue if you still can reproduce it, thanks.

closed 2017-11-30 by bafu