DT42/BerryNet #12: No log errors, but inference fails (x.jpg.done is 0 bytes).  Status shows 2 services failed.

TheMosquito, 2017-11-15:
I would like to try using BerryNet on my Raspberry Pi 3, but two of the services listed in the berrynet-manager status output are failing and I am unclear about how to debug this.  I have tried using both the hardware raspi-cam, and a USB camera.  I am able to capture stills and stream from both of these cameras

The MQTT log shows chunks like this that seem to look successful:

[2017-11-15 23:29:54] camera client: publishing image.
[2017-11-15 23:29:54] camera client: on topic berrynet/event/camera, received message snapshot_boardcam.
[2017-11-15 23:29:50] inference client: saved buffer to image /usr/local/berrynet/inference/image/snapshot-20171115-232950.jpg successfully.
[2017-11-15 23:29:50] inference client: on topic berrynet/action/inference, received 276635 bytes.
[2017-11-15 23:29:50] Image done file /usr/local/berrynet/inference/image/snapshot-20171115-232950.jpg.done is ready.

The image file referenced above has the size that matches what the inference client receives:

pi@rpiai:~ $ ls -l /usr/local/berrynet/inference/image/snapshot-20171115-232950.jpg
-rw-r--r-- 1 root pi 276635 Nov 15 23:29 /usr/local/berrynet/inference/image/snapshot-20171115-232950.jpg
pi@rpiai:~ $

I can manually view that image file and it looks good.

The inference client gives no indication of trouble in the log, but the output file that it names above is always 0 bytes:

pi@rpiai:~ $ ls -l /usr/local/berrynet/inference/image/snapshot-20171115-232950.jpg.done
-rw-r--r-- 1 root pi 0 Nov 15 23:29 /usr/local/berrynet/inference/image/snapshot-20171115-232950.jpg.done
pi@rpiai:~ $ 

So something is failing at the inference stage.  When I look at the web client I see no images in camera snapshot or the inference result panes, just black backgrounds.

The berrynet-manager status shows 2 failed services (detection_fast_server.service, and line.service):

pi@rpiai:~/BerryNet $ berrynet-manager status
● detection_fast_server.service - detection server
   Loaded: loaded (/etc/systemd/system/detection_fast_server.service; disabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Wed 2017-11-15 23:37:35 UTC; 5s ago
  Process: 10002 ExecStart=/bin/bash utils/localrun.sh /usr/local/berrynet/inference/image (code=exited, status=1/FAILURE)
 Main PID: 10002 (code=exited, status=1/FAILURE)
Nov 15 23:37:35 rpiai systemd[1]: detection_fast_server.service: Unit entered failed state.
Nov 15 23:37:35 rpiai systemd[1]: detection_fast_server.service: Failed with result 'exit-code'.
● agent.service - MQTT client agent for classifier or detector
   Loaded: loaded (/etc/systemd/system/agent.service; disabled; vendor preset: enabled)
   Active: active (running) since Wed 2017-11-15 23:17:17 UTC; 20min ago
 Main PID: 8120 (node)
   CGroup: /system.slice/agent.service
           └─8120 /usr/bin/node agent.js
Nov 15 23:29:50 rpiai node[8120]: rename event for snapshot-20171115-232950.jpg, but it is not inference result done file.
Nov 15 23:29:50 rpiai node[8120]: change event for snapshot-20171115-232950.jpg, ignore it.
Nov 15 23:29:50 rpiai node[8120]: inference client: saved buffer to image /usr/local/berrynet/inference/image/snapshot-20171115-23
Nov 15 23:29:54 rpiai node[8120]: inference client: on topic berrynet/action/inference, received 255160 bytes.
Nov 15 23:29:54 rpiai node[8120]: Image done file /usr/local/berrynet/inference/image/snapshot-20171115-232954.jpg.done is ready.
Nov 15 23:29:54 rpiai node[8120]: rename event for snapshot-20171115-232954.jpg, but it is not inference result done file.
Nov 15 23:29:54 rpiai node[8120]: rename event for snapshot-20171115-232954.jpg.done, but it is not inference result done file.
Nov 15 23:29:54 rpiai node[8120]: rename event for snapshot-20171115-232954.jpg.done, but it is not inference result done file.
Nov 15 23:29:54 rpiai node[8120]: change event for snapshot-20171115-232954.jpg, ignore it.
Nov 15 23:29:54 rpiai node[8120]: inference client: saved buffer to image /usr/local/berrynet/inference/image/snapshot-20171115-23
● broker.service - MQTT broker
   Loaded: loaded (/etc/systemd/system/broker.service; disabled; vendor preset: enabled)
   Active: active (running) since Wed 2017-11-15 23:17:17 UTC; 20min ago
 Main PID: 8119 (node)
   CGroup: /system.slice/broker.service
           └─8119 /usr/bin/node broker.js
Nov 15 23:29:54 rpiai node[8119]: Server published. Avoid showing package content here.
Nov 15 23:29:54 rpiai node[8119]: Server published. Avoid showing package content here.
Nov 15 23:29:54 rpiai node[8119]: Server published. Avoid showing package content here.
Nov 15 23:29:54 rpiai node[8119]: Server published. Avoid showing package content here.
Nov 15 23:29:54 rpiai node[8119]: Server published. Avoid showing package content here.
Nov 15 23:29:54 rpiai node[8119]: Server published. Avoid showing package content here.
Nov 15 23:29:54 rpiai node[8119]: Server published. Avoid showing package content here.
Nov 15 23:29:54 rpiai node[8119]: Server published. Avoid showing package content here.
Nov 15 23:32:36 rpiai node[8119]: Server published. Avoid showing package content here.
Nov 15 23:32:36 rpiai node[8119]: Server published. Avoid showing package content here.
● dashboard.service - Dashboard webserver
   Loaded: loaded (/etc/systemd/system/dashboard.service; disabled; vendor preset: enabled)
[... 47 lines cut ...]
   Loaded: loaded (/etc/systemd/system/data_collector.service; disabled; vendor preset: enabled)
   Active: active (running) since Wed 2017-11-15 23:17:17 UTC; 20min ago
 Main PID: 8114 (node)
   CGroup: /system.slice/data_collector.service
           └─8114 /usr/bin/node data_collector.js
Nov 15 23:29:50 rpiai node[8114]: 2017-11-15T23:29:50.054Z
Nov 15 23:29:50 rpiai node[8114]: log client: saved buffer to image successfully.
Nov 15 23:29:50 rpiai node[8114]: Unsubscribed topic berrynet/action/log
Nov 15 23:29:50 rpiai node[8114]: Unsubscribed topic berrynet/action/log
Nov 15 23:29:54 rpiai node[8114]: Unsubscribed topic berrynet/action/log
Nov 15 23:29:54 rpiai node[8114]: Unsubscribed topic berrynet/action/log
Nov 15 23:29:54 rpiai node[8114]: Get berrynet/action/inference
Nov 15 23:29:54 rpiai node[8114]: 2017-11-15T23:29:54.509Z
Nov 15 23:29:54 rpiai node[8114]: log client: saved buffer to image successfully.
Nov 15 23:29:54 rpiai node[8114]: Unsubscribed topic berrynet/action/log
● line.service - LINE client agent for notification
   Loaded: loaded (/etc/systemd/system/line.service; disabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Wed 2017-11-15 23:37:36 UTC; 4s ago
  Process: 9996 ExecStart=/usr/bin/node line.js (code=exited, status=1/FAILURE)
 Main PID: 9996 (code=exited, status=1/FAILURE)
Nov 15 23:37:36 rpiai systemd[1]: line.service: Unit entered failed state.
Nov 15 23:37:36 rpiai systemd[1]: line.service: Failed with result 'exit-code'.
pi@rpiai:~/BerryNet $ 

Maybe a darknet issue?

bafu, 2017-11-17:
@TheMosquito Thank you for trying BerryNet and reporting the issue report. I will check it and share my investigation result with you soon.

bafu, 2017-11-30:
@TheMosquito For the detection_fast_server.service error, it should be fixed in [v2.1.1](https://github.com/DT42/BerryNet/releases/tag/2.1.1); for the line.service error, it is because LINE client is enabled by default, but the default config values are empty in config.js. We will disable these additional clients and provide more tutorials about them in the next release.

I will keep this issue open until the line.service error is really resolved.

WeberChen, 2018-01-02:
@bafu  Ignore the crash of configure step, I just run the inference step and also meet the same problem as detection_fast_server.service error. The log information is the same as @TheMosquito 's first problem in both versions of v2.2.0 and v2.1.1. The OS I used is Ubuntu-Mate. Could you give me some suggestion? Thanks.

TheMosquito, 2018-02-13:
@bafu: I have left this alone for the last almost 3 months, and tried again today on the current Raspbian base image and the current BerryNet.  I got errors related to ZMQ not being installed so I installed that.  Now I am seeing the same error that was reported in issue #22.  I also got many other errors relating to configuration omissions, but I think I have fixed those errors.  It would be good if you would provide reasonable "default" values for configuration or clearly state what configuration must be provided to make it work.  I feel that there is a lot of information missing in your documentation.  Would it be possible for someone to provide a concise and *reliable* series of steps to flash an appropriate base Linux image (if you need a specific distro or version, please state this), and install BerryNet without error (even with the above issue and many errors during my ./configure step, the final install message says, "successful", but I don't think it was successful) and then configure correctly all the needed variables in config.js (what is needed there), and then finally run "barrynet-manager start" such that "berrynet-manager status" shows all of the services up and running without error?  I think other people would find this information useful too.

bafu, 2018-07-09:
@TheMosquito @WeberChen The [v2.3.0 installation image](https://github.com/DT42/BerryNet/releases/tag/v2.3.0) is available now, and it will save community friends' efforts to build from scratch. Please try it and feel free to let me know if there is any issues, thanks!

closed 2018-07-09 by bafu