DT42/BerryNet #13: nothing visible in the webpage; errors at launch

adioanca, 2017-11-23:
Hi,
this is my syslog:
---
Nov 23 12:11:15 raspberrypi systemd[1]: Started detection server.
Nov 23 12:11:15 raspberrypi bash[3984]: SNAPSHOT_DIR: /usr/local/berrynet/inference/image
Nov 23 12:11:15 raspberrypi systemd[1]: Started MQTT client agent for classifier or detector.
Nov 23 12:11:15 raspberrypi systemd[1]: Started LINE client agent for notification.
Nov 23 12:11:15 raspberrypi systemd[1]: Started MQTT client agent for DL inference data collection..
Nov 23 12:11:15 raspberrypi systemd[1]: Started MQTT client agent for logging..
Nov 23 12:11:15 raspberrypi systemd[1]: Started MQTT client to get snapshot from Pi camera or IP camera..
Nov 23 12:11:15 raspberrypi systemd[1]: Started MQTT client to detect local image event.
Nov 23 12:11:15 raspberrypi systemd[1]: Started Dashboard webserver.
Nov 23 12:11:15 raspberrypi systemd[1]: Started MQTT broker.
Nov 23 12:11:15 raspberrypi bash[3984]: [1511439075.737] watch_dir:  /usr/local/berrynet/inference/image
Nov 23 12:11:15 raspberrypi bash[3984]: [1511439075.737] pid:  /tmp/detectord.pid
Nov 23 12:11:15 raspberrypi bash[3984]: Traceback (most recent call last):
Nov 23 12:11:15 raspberrypi bash[3984]:   File "detectord.py", line 137, in <module>
Nov 23 12:11:15 raspberrypi bash[3984]:     stderr=subprocess.STDOUT)
Nov 23 12:11:15 raspberrypi bash[3984]:   File "/usr/lib/python2.7/subprocess.py", line 390, in __init__
Nov 23 12:11:15 raspberrypi bash[3984]:     errread, errwrite)
Nov 23 12:11:15 raspberrypi bash[3984]:   File "/usr/lib/python2.7/subprocess.py", line 1024, in _execute_child
Nov 23 12:11:15 raspberrypi bash[3984]:     raise child_exception
Nov 23 12:11:15 raspberrypi bash[3984]: OSError: [Errno 2] No such file or directory
Nov 23 12:11:15 raspberrypi systemd[1]: detection_fast_server.service: Main process exited, code=exited, status=1/FAILURE
Nov 23 12:11:15 raspberrypi systemd[1]: detection_fast_server.service: Unit entered failed state.
Nov 23 12:11:15 raspberrypi systemd[1]: detection_fast_server.service: Failed with result 'exit-code'.
Nov 23 12:11:16 raspberrypi node[3997]: Dashboard Server running as PORT 8080
Nov 23 12:11:16 raspberrypi node[3997]: Press CTRL + C to stop
Nov 23 12:11:19 raspberrypi node[3993]: Failed to create data storage dir.
Nov 23 12:11:20 raspberrypi node[3992]: /usr/local/berrynet/node_modules/@line/bot-sdk/dist/client.js:9
Nov 23 12:11:20 raspberrypi node[3992]:             throw new Error("no channel access token");
Nov 23 12:11:20 raspberrypi node[3992]:             ^
Nov 23 12:11:20 raspberrypi node[3992]: Error: no channel access token
Nov 23 12:11:20 raspberrypi node[3992]:     at new Client (/usr/local/berrynet/node_modules/@line/bot-sdk/dist/client.js:9:19)
Nov 23 12:11:20 raspberrypi node[3992]:     at Object.<anonymous> (/usr/local/berrynet/line.js:39:20)
Nov 23 12:11:20 raspberrypi node[3992]:     at Module._compile (module.js:570:32)
Nov 23 12:11:20 raspberrypi node[3992]:     at Object.Module._extensions..js (module.js:579:10)
Nov 23 12:11:20 raspberrypi node[3992]:     at Module.load (module.js:487:32)
Nov 23 12:11:20 raspberrypi node[3992]:     at tryModuleLoad (module.js:446:12)
Nov 23 12:11:20 raspberrypi node[3992]:     at Function.Module._load (module.js:438:3)
Nov 23 12:11:20 raspberrypi node[3992]:     at Module.runMain (module.js:604:10)
Nov 23 12:11:20 raspberrypi node[3992]:     at run (bootstrap_node.js:383:7)
Nov 23 12:11:20 raspberrypi node[3992]:     at startup (bootstrap_node.js:149:9)
Nov 23 12:11:20 raspberrypi systemd[1]: line.service: Main process exited, code=exited, status=1/FAILURE
Nov 23 12:11:20 raspberrypi systemd[1]: line.service: Unit entered failed state.
Nov 23 12:11:20 raspberrypi systemd[1]: line.service: Failed with result 'exit-code'.
---

Traced it down to: /home/pi/BerryNet/utils/darknet/detectord.py:
    cmd = ['./darknet', 'detector', 'test', data, cfg, weight, '-out', '/usr/local/berrynet/dashboard/www/freeboard/snapshot']
    darknet = subprocess.Popen(cmd, bufsize=0,
            stdin=subprocess.PIPE,
            stderr=subprocess.STDOUT)
---

I've searched through all my RPi /, there is no "./darknet" executable. Could you explain a bit what's going on?

Thank you

adioanca, 2017-11-23:
This is because ./patch/01-detection-backend.patch is not applied correctly.
Probably darknet code changed since you've created that patch.

Maybe you can find some APIs to use instead of changing darknet code by patches... ?

bafu, 2017-11-30:
@adioanca Thank you for reporting this issue, and it has been fixed by [v2.1.1](https://github.com/DT42/BerryNet/releases/tag/2.1.1).

closed 2017-11-30 by bafu