DT42/BerryNet #16: Raspberry Pi 3 Crash issue

Allwin473, 2017-12-05:
Hi,
I have faced an Raspberry Pi 3 board crashing issue, while running ./configure in my setup.
I find every time, it is crashing while installing convolution module (53/145) of  dark flow package.
Can you give me your input on this?

bafu, 2017-12-14:
@Allwin473 I can help check the installation issue. Can you share the installation log (`/tmp/berrynet.log`) and your BerryNet version with me?

Allwin473, 2017-12-15:
[berrynet.log](https://github.com/DT42/BerryNet/files/1562224/berrynet.log)
Hi Bafu,
 I have attached berrynet.log file with this.After this 53 rd Installation, Raspberry Pi 3 is totally crashed. Mouse pointer is not working properly.

haodan1989, 2017-12-21:
I'm having the same issue running the configure, but in my case, the crash happened on the 50th installation, tried 4,5 times doesn't work.

WeberChen, 2017-12-28:
I'm also having the same issue many times while running the configure on the 50th step of cloning the darknet-nnpack: [50/145] CC src/ref/relu-input-gradient.c . Even I closed the other installation steps, the crash still happened. By the way, my RPi3-OS is Ubuntu Mate. Thanks.

Allwin473, 2018-01-18:
Hi All,
  Thanks for your support on this issue. I have done some modification in the configure file to make it work.
1.  While installing confu (Same version mentioned in configuration file), i downloaded it and installed it from locally. 
2.  As i kept my code in system memory, Some Copy commands are not working properly as it doesn't have permission. I have added sudo with them to make it work. 
3. As my Raspi 3 doesn't have much processing power to run this, it hangs and halted for more time. If you are facing the same issue, try with the below workaround.
Work around i used:
-------------------------
   1) Try to stop all background services ( Go to Rasrberry Pi configuration and disable all Interface running)
 (Note: If you are using putty, you can't access raspberry pi3 as you disable all interfaces (if SSH Interface is disable ))
   2) After giving sudo ./configure command,remove your Mouse, Keyboard or Monitor from your raspberry Pi 3 as it takes some processing. This installation may take 15 to 30 minutes, wait for it to complete.
   3) It will work with probability of 2/3. To make sure installation is going on, Monitor your ACT LED indicatorin your RPi3(It should blink). If it is glowing continuously, It is hanging somewhere. So try to restart your Instatllation.

closed 2018-01-18 by Allwin473