DT42/BerryNet #17: All services reporting failure, code=exited, status=8, service entered failed state.

haodan1989, 2017-12-24:
Step1: Clone the repo
Step2: ./configure
Step3: berrynet-manager start
Step4: berrynet-manager status

The log on the RP3 shows only detection_fast_server.service is on, and all other services are down. something I noticed during the configure running: even though it reported success in the end, there are multiple warnings generated in the middle, so is it considered fully finished or there is something I'm missing that caused the issue ?

gianlucabreda, 2018-01-07:
I am experiencing the same issue as his (no service running, status=8). Also had some errors during the installation, some disappeared after updating and upgrading but some others remained. Received a successful install message as well.

tomwu48, 2018-05-11:
I see the same issue when installing BerryNet on my RaspBerry Pi3. I did some investigation. I think the problem is when executing this line "/usr/bin/node agent.js", the following error is reported.
/usr/local/berrynet/interfence/agent.js:20
const fs=require('fs')
^^^^^
Syntax Error: Use of const in strict mode.
When I type "/usr/bin/node -v", I see "v0.10.29". I guess node.js is not installed properly.

tomwu48, 2018-05-11:
When using the configure script to install nodejs, this command "apt-get update" will be executed. It will report error message "bad header line" when getting something from https://github.com/DT42/BerryNet-repo/blob/master/dists/stretch/main/binary-armhf/Packages. The URL is from apt/berrynet.list. So I comment out everything in berrynet.list and do configure again. The issue can be solved. The apt tool I'm using is version 1.0.9.8.4. Perhaps it is too old and cannot work with the Packages file from BerryNet-repo.

bafu, 2018-07-09:
@haodan1989 @gianlucabreda @tomwu48 The [v2.3.0 installation image](https://github.com/DT42/BerryNet/releases/tag/v2.3.0) is available now, and it will save community friends' efforts to build from scratch. Please try it and feel free to let me know if there is any issues, thanks!

closed 2018-07-09 by bafu