DT42/BerryNet #23: N: See apt-secure(8) manpage for repository creation and user configuration details.

Tondy, 2018-02-20:
Hello

I need some help as I am receiving the error below every time I am running sudo apt-get update. 
Please help

W: GPG error: https://raw.githubusercontent.com/DT42/BerryNet-repo/master stretch InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY A3211A87D7ACFE23
E: The repository 'https://raw.githubusercontent.com/DT42/BerryNet-repo/master stretch InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

tuansacer, 2018-03-20:
I'm have the same error :(

bafu, 2018-04-03:
@Tondy @tuansacer Sorry for hitting the issue because you find the easter egg!

We're working on Python-based BerryNet which has not been announced yet.

To resolve this issue, please use this command below:

```
$ sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys A3211A87D7ACFE23
$ sudo apt update
```

bafu, 2018-07-08:
This issue has been fixed in v2.3.0.

closed 2018-07-08 by bafu

hemendram0, 2019-02-05:
i am facing same problem. help me

muketot, 2019-03-21:
Pls help me

bandarikanth, 2019-08-28:
remove sources .list and index.html in /etc/apt dir

# cp /etc/apt/preferences.d/index.html /tmp
# rm -rf /etc/apt/preferences.d/index.html
#cp /etc/apt/sources.list  /tmp
# rm -rf sources.list

     then reboot

PrinsZuko, 2020-01-17:
> @Tondy @tuansacer Sorry for hitting the issue because you find the easter egg!
> 
> We're working on Python-based BerryNet which has not been announced yet.
> 
> To resolve this issue, please use this command below:
> 
> ```
> $ sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys A3211A87D7ACFE23
> $ sudo apt update
> ```
> 
> I will keep this issue open until Python-based BerryNet is announced officially.

Each time I try these type of fixes it just says "command not found" in debian 9.5

grandpaul, 2020-01-19:
Hi PrinsZuko, 

Can you check if sudo is installed? apt-key and apt must be run as root. 
If your system admin doesn't install "sudo" package, please install it first.