DT42/BerryNet #25: Install BerryNet : Can you provide and .img file for raspberry PI3B

angenieux, 2018-04-03:
Dear All,
thanks for your huge job !
In order to go faster in tests, can you provide an .img file of a pre-configured BerryNet platform for PI3B and Rapberry pi official camera V2 ?
Burning an image on a SD card is so faster than compiling all the dependencies and module.

Best regards
Nicolas Angenieux

bafu, 2018-04-18:
@angenieux Thank you for the suggestion. As you mentioned, because many people met similar installation issues, we are planning to deliver system image in the next release, and improve the Wiki pages w/ detailed compilation steps for the users who want to install from scratch by themselves.

bafu, 2018-07-09:
@angenieux The [v2.3.0 installation image](https://github.com/DT42/BerryNet/releases/tag/v2.3.0) is available now, and it will save community friends' efforts to build from scratch. Please try it and feel free to let me know if there is any issues, thanks!

closed 2018-07-09 by bafu