DT42/BerryNet #26: not able to get detected image on web ui

codehArt, 2018-04-19:
When I run 
mosquitto_pub -h localhost -t berrynet/event/localImage -m data/person.jpg
it is publishing data and also checked mongoDB entries it is contioniously increasing,
unfortunately I am not able to view detected data 
I am getting this error on
berrynet-manager log 

-- Subject: Unit line.service has finished shutting down
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- Unit line.service has finished shutting down.
Apr 19 19:53:36 raspberrypi systemd[1]: Started LINE client agent for notification.
-- Subject: Unit line.service has finished start-up
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- Unit line.service has finished starting up.
-- 
-- The start-up result is done.
Apr 19 19:53:38 raspberrypi node[3864]: /usr/local/berrynet/node_modules/@line/bot-sdk/dist/client.js:9
Apr 19 19:53:38 raspberrypi node[3864]:             throw new Error("no channel access token");
Apr 19 19:53:38 raspberrypi node[3864]:             ^
Apr 19 19:53:38 raspberrypi node[3864]: Error: no channel access token
Apr 19 19:53:38 raspberrypi node[3864]:     at new Client (/usr/local/berrynet/node_modules/@line/bot-sdk/dist/client.js:9:19)
Apr 19 19:53:38 raspberrypi node[3864]:     at Object.<anonymous> (/usr/local/berrynet/line.js:39:20)
Apr 19 19:53:38 raspberrypi node[3864]:     at Module._compile (module.js:577:32)
Apr 19 19:53:38 raspberrypi node[3864]:     at Object.Module._extensions..js (module.js:586:10)
Apr 19 19:53:38 raspberrypi node[3864]:     at Module.load (module.js:494:32)
Apr 19 19:53:38 raspberrypi node[3864]:     at tryModuleLoad (module.js:453:12)
Apr 19 19:53:38 raspberrypi node[3864]:     at Function.Module._load (module.js:445:3)
Apr 19 19:53:38 raspberrypi node[3864]:     at Module.runMain (module.js:611:10)
Apr 19 19:53:38 raspberrypi node[3864]:     at run (bootstrap_node.js:387:7)
Apr 19 19:53:38 raspberrypi node[3864]:     at startup (bootstrap_node.js:153:9)
Apr 19 19:53:38 raspberrypi systemd[1]: line.service: Main process exited, code=exited, status=1/FAILURE
Apr 19 19:53:38 raspberrypi systemd[1]: line.service: Unit entered failed state.
Apr 19 19:53:38 raspberrypi systemd[1]: line.service: Failed with result 'exit-code'.

hale, 2018-05-06:
I think that error might be because of LINE not being configured? 

I don’t have this project installed yet so this is just from browsing the repo.

bafu, 2018-07-09:
@codehArt The [v2.3.0 installation image](https://github.com/DT42/BerryNet/releases/tag/v2.3.0) is available now, and it will save community friends' efforts to build from scratch. Please try it and feel free to let me know if there is any issues, thanks!

closed 2018-07-09 by bafu