DT42/BerryNet #30: Raspberry Pi3+ Install Doesn't Work

jeylites, 2019-04-20:
Hello 

Thanks for doing this project. I'm trying to get this to work on a Raspberry Pi3 B+ but I can't seem to get it working. The browser is not showing up. http://192.168.1.201:8080/index.html#source=dashboard.json

The installation I followed was [Steps for JavaScript Codebase](https://github.com/DT42/BerryNet/wiki/Installation#steps-for-javascript-codebase)

Here are the steps to install BerryNet on RPi:

Get BerryNet source codes
```
$ git clone https://github.com/DT42/BerryNet.git
$ cd BerryNet
```
Execute installation script

```
$ ./configure
```

```
Get:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease [15.0 kB]
Get:2 http://archive.raspberrypi.org/debian stretch InRelease [25.4 kB]
Get:3 http://raspbian.raspberrypi.org/raspbian stretch/main armhf Packages [11.7 MB]
Get:4 http://archive.raspberrypi.org/debian stretch/main armhf Packages [223 kB]
Get:5 http://archive.raspberrypi.org/debian stretch/ui armhf Packages [44.9 kB]
Fetched 12.0 MB in 14s (839 kB/s)
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
curl is already the newest version (7.52.1-5+deb9u9).
git is already the newest version (1:2.11.0-3+deb9u4).
lsb-release is already the newest version (9.20161125+rpi1).
lsb-release set to manually installed.
python3-dev is already the newest version (3.5.3-1).
python3-dev set to manually installed.
wget is already the newest version (1.18-5+deb9u3).
python3-pip is already the newest version (9.0.1-2+rpt2).
The following additional packages will be installed:
  comerr-dev ghostscript gsfonts imagemagick-6-common imagemagick-6.q16
  krb5-multidev libboost-filesystem1.58.0 libboost-program-options1.58.0
  libboost-system1.58.0 libboost-thread1.58.0 libboost-thread1.62.0 libc-ares2
  libev4 libgs9 libgs9-common libgssrpc4 libjxr-tools libjxr0
  libkadm5clnt-mit11 libkadm5srv-mit11 libkdb5-8 liblqr-1-0
  libmagickcore-6.q16-3 libmagickcore-6.q16-3-extra libmagickwand-6.q16-3
  libmongoclient-dev libmongoclient0 libmosquitto1 libnetpbm10 libpcap0.8
  libpcrecpp0v5 libuv1 libv8-3.14.5 libwebsockets8 libwmf0.2-7 libyaml-0-2
  mongodb-clients mongodb-dev mongodb-server netpbm
Suggested packages:
  doc-base ghostscript-x imagemagick-doc autotrace cups-bsd | lpr | lprng
  enscript gimp gnuplot grads graphviz hp2xx html2ps libwmf-bin mplayer povray
  radiance sane-utils texlive-base-bin transfig ufraw-batch krb5-doc krb5-user
  inkscape libyaml-doc apparmor
The following NEW packages will be installed:
  comerr-dev fswebcam ghostscript gsfonts imagemagick imagemagick-6-common
  imagemagick-6.q16 krb5-multidev libboost-filesystem1.58.0
  libboost-program-options1.58.0 libboost-system1.58.0 libboost-thread1.58.0
  libboost-thread1.62.0 libc-ares2 libev4 libgssrpc4 libjxr-tools libjxr0
  libkadm5clnt-mit11 libkadm5srv-mit11 libkdb5-8 libkrb5-dev liblqr-1-0
  libmagickcore-6.q16-3 libmagickcore-6.q16-3-extra libmagickwand-6.q16-3
  libmongoclient-dev libmongoclient0 libmosquitto1 libnetpbm10 libpcap0.8
  libpcrecpp0v5 libuv1 libv8-3.14.5 libwebsockets8 libwmf0.2-7 libyaml-0-2
  libyaml-dev libzmq3-dev mongodb mongodb-clients mongodb-dev mongodb-server
  mosquitto mosquitto-clients netpbm
The following packages will be upgraded:
  libgs9 libgs9-common
2 upgraded, 46 newly installed, 0 to remove and 19 not upgraded.
Need to get 51.2 MB of archives.
After this operation, 147 MB of additional disk space will be used.
Get:1 http://archive.raspberrypi.org/debian stretch/main armhf libc-ares2 armhf 1.14.0-1~bpo9+1 [80.7 kB]
[... 49 lines cut ...]
Extracting templates from packages: 62%
Extracting templates from packages: 100%
Fetched 51.2 MB in 22s (2,321 kB/s)
Selecting previously unselected package liblqr-1-0:armhf.
(Reading database ... 
(Reading database ... 5%
(Reading database ... 10%
(Reading database ... 15%
(Reading database ... 20%
(Reading database ... 25%
(Reading database ... 30%
(Reading database ... 35%
(Reading database ... 40%
(Reading database ... 45%
(Reading database ... 50%
(Reading database ... 55%
(Reading database ... 60%
(Reading database ... 65%
(Reading database ... 70%

```

Hardware: Raspberry Pi3 B+
Just a note:
I have MQTT broker installed on a different RPI3 B+ running my home automation.

bafu, 2019-04-20:
@jeylites Thank you for reporting this issue. I will check it and share the investigation result with you.

Before the result is available, you can try the Python codebase (install Debian packages) and here is the beta version of [Python codebase example usage](https://github.com/DT42/BerryNet/tree/feature-update-default-example).

If you need more real-time discussion, welcome to join the Telegram group: https://t.me/berrynetdev

jeylites, 2019-04-20:
@bafu 

I'm testing the Python codebase as you suggested. The first two lines that I ran has been successful but the next I get the following.  What is Wheel and where should I get it?
```

pi@raspberrypi:~ $ python3 setup.py bdist_wheel
python3: can't open file 'setup.py': [Errno 2] No such file or directory
pi@raspberrypi:~ $ 

```

jeylites, 2019-04-20:
@bafu 

Please ignore the top, I ran the following was successful but came to ` supervisorctl`I get : command not found

```
$ git clone https://github.com/DT42/BerryNet.git
$ cd BerryNet
$ ./configure 
```

```
pi@raspberrypi:~ $ git clone https://github.com/DT42/BerryNet.git
fatal: destination path 'BerryNet' already exists and is not an empty directory.
pi@raspberrypi:~ $ sudo supervisorctl status all
sudo: supervisorctl: command not found
pi@raspberrypi:~ $ supervisorctl
-bash: supervisorctl: command not found
pi@raspberrypi:~ $ cd BerryNet
pi@raspberrypi:~/BerryNet $ ./configure
Hit:1 http://archive.raspberrypi.org/debian stretch InRelease
Get:3 http://raspbian.raspberrypi.org/raspbian stretch InRelease [15.0 kB]
Hit:4 https://deb.nodesource.com/node_10.x stretch InRelease
Hit:2 http://repo.berrynet.org.s3-website-us-east-1.amazonaws.com stretch/ InRelease
Fetched 15.0 kB in 3s (4,647 B/s)
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
curl is already the newest version (7.52.1-5+deb9u9).
fswebcam is already the newest version (20140113-1).
git is already the newest version (1:2.11.0-3+deb9u4).
imagemagick is already the newest version (8:6.9.7.4+dfsg-11+deb9u6).
libkrb5-dev is already the newest version (1.15-1+deb9u1).
libyaml-dev is already the newest version (0.1.7-2).
libzmq3-dev is already the newest version (4.2.1-4+deb9u1).
lsb-release is already the newest version (9.20161125+rpi1).
mongodb is already the newest version (1:2.4.14-4).
mosquitto is already the newest version (1.4.10-3+deb9u4).
mosquitto-clients is already the newest version (1.4.10-3+deb9u4).
python3-dev is already the newest version (3.5.3-1).
wget is already the newest version (1.18-5+deb9u3).
python3-pip is already the newest version (9.0.1-2+rpt2).
0 upgraded, 0 newly installed, 0 to remove and 19 not upgraded.
Collecting cython
  Using cached https://files.pythonhosted.org/packages/f8/da/c979464858b257b21a6472a85285548c91f5b4dc773cb049cfdfb3ceeb02/Cython-0.29.7.tar.gz
Collecting logzero
  Using cached https://files.pythonhosted.org/packages/97/24/27295d318ea8976b12cf9cc51d82e7c7129220f6a3cc9e3443df3be8afdb/logzero-1.5.0-py2.py3-none-any.whl
Collecting paho-mqtt
  Downloading https://www.piwheels.org/simple/paho-mqtt/paho_mqtt-1.4.0-py3-none-any.whl (48kB)
Collecting watchdog
  Downloading https://www.piwheels.org/simple/watchdog/watchdog-0.9.0-py3-none-any.whl (67kB)
Collecting pathtools>=0.1.1 (from watchdog)
  Downloading https://www.piwheels.org/simple/pathtools/pathtools-0.1.2-py3-none-any.whl
Collecting argh>=0.24.1 (from watchdog)
Exception:
Traceback (most recent call last):
  File "/usr/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connectionpool.py", line 594, in urlopen
    chunked=chunked)
  File "/usr/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connectionpool.py", line 391, in _make_request
    six.raise_from(e, None)
  File "<string>", line 2, in raise_from
  File "/usr/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connectionpool.py", line 387, in _make_request
    httplib_response = conn.getresponse()
  File "/usr/lib/python3.5/http/client.py", line 1198, in getresponse
    response.begin()
  File "/usr/lib/python3.5/http/client.py", line 297, in begin
    version, status, reason = self._read_status()
  File "/usr/lib/python3.5/http/client.py", line 266, in _read_status
    raise RemoteDisconnected("Remote end closed connection without"
http.client.RemoteDisconnected: Remote end closed connection without response
[... 139 lines cut ...]
libdarknet is already the newest version (20171109-3).
0 upgraded, 0 newly installed, 0 to remove and 19 not upgraded.

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

Reading package lists...
Building dependency tree...
Reading state information...
tinyyolovoc is already the newest version (20170816).
0 upgraded, 0 newly installed, 0 to remove and 19 not upgraded.
Installation is completed successfully!

Please send /home/pi/BerryNet/berrynet.log and /usr/local/berrynet/npm-debug.log
to https://groups.google.com/a/dt42.io/d/forum/berrynet
if there is any issue.
pi@raspberrypi:~/BerryNet $ sudo supervisorctl status all
sudo: supervisorctl: command not found
pi@raspberrypi:~/BerryNet $ sudo supervisorctl stop camera
sudo: supervisorctl: command not found
pi@raspberrypi:~/BerryNet $ sudo supervisorctl restart all
sudo: supervisorctl: command not found
pi@raspberrypi:~/BerryNet $ sudo supervisorctl tail camera stderr
sudo: supervisorctl: command not found
pi@raspberrypi:~/BerryNet $ 
```

jeylites, 2019-04-20:
@bafu 

I decided to start from scratch. I followed from the previous post still can't get this going. For the BerryNet v3 install I when to as link   [Installation](https://github.com/DT42/BerryNet/tree/feature-update-default-example)

And coped the following to terminal...

```
$ git clone https://github.com/DT42/BerryNet.git
$ cd BerryNet
$ ./configure
```

LOGS 

```
Get:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease [15.0 kB]
Get:2 http://archive.raspberrypi.org/debian stretch InRelease [25.4 kB]
Get:3 http://raspbian.raspberrypi.org/raspbian stretch/main armhf Packages [11.7 MB]
Get:4 http://archive.raspberrypi.org/debian stretch/main armhf Packages [223 kB]
Get:5 http://archive.raspberrypi.org/debian stretch/ui armhf Packages [44.9 kB]
Fetched 12.0 MB in 16s (718 kB/s)
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
curl is already the newest version (7.52.1-5+deb9u9).
git is already the newest version (1:2.11.0-3+deb9u4).
lsb-release is already the newest version (9.20161125+rpi1).
lsb-release set to manually installed.
python3-dev is already the newest version (3.5.3-1).
python3-dev set to manually installed.
wget is already the newest version (1.18-5+deb9u3).
python3-pip is already the newest version (9.0.1-2+rpt2).
The following additional packages will be installed:
  comerr-dev ghostscript gsfonts imagemagick-6-common imagemagick-6.q16
  krb5-multidev libboost-filesystem1.58.0 libboost-program-options1.58.0
  libboost-system1.58.0 libboost-thread1.58.0 libboost-thread1.62.0 libc-ares2
  libev4 libgs9 libgs9-common libgssrpc4 libjxr-tools libjxr0
  libkadm5clnt-mit11 libkadm5srv-mit11 libkdb5-8 liblqr-1-0
  libmagickcore-6.q16-3 libmagickcore-6.q16-3-extra libmagickwand-6.q16-3
  libmongoclient-dev libmongoclient0 libmosquitto1 libnetpbm10 libpcap0.8
  libpcrecpp0v5 libuv1 libv8-3.14.5 libwebsockets8 libwmf0.2-7 libyaml-0-2
  mongodb-clients mongodb-dev mongodb-server netpbm
Suggested packages:
  doc-base ghostscript-x imagemagick-doc autotrace cups-bsd | lpr | lprng
  enscript gimp gnuplot grads graphviz hp2xx html2ps libwmf-bin mplayer povray
  radiance sane-utils texlive-base-bin transfig ufraw-batch krb5-doc krb5-user
  inkscape libyaml-doc apparmor
The following NEW packages will be installed:
  comerr-dev fswebcam ghostscript gsfonts imagemagick imagemagick-6-common
  imagemagick-6.q16 krb5-multidev libboost-filesystem1.58.0
  libboost-program-options1.58.0 libboost-system1.58.0 libboost-thread1.58.0
  libboost-thread1.62.0 libc-ares2 libev4 libgssrpc4 libjxr-tools libjxr0
  libkadm5clnt-mit11 libkadm5srv-mit11 libkdb5-8 libkrb5-dev liblqr-1-0
  libmagickcore-6.q16-3 libmagickcore-6.q16-3-extra libmagickwand-6.q16-3
  libmongoclient-dev libmongoclient0 libmosquitto1 libnetpbm10 libpcap0.8
  libpcrecpp0v5 libuv1 libv8-3.14.5 libwebsockets8 libwmf0.2-7 libyaml-0-2
  libyaml-dev libzmq3-dev mongodb mongodb-clients mongodb-dev mongodb-server
  mosquitto mosquitto-clients netpbm
The following packages will be upgraded:
  libgs9 libgs9-common
2 upgraded, 46 newly installed, 0 to remove and 19 not upgraded.
Need to get 51.2 MB of archives.
After this operation, 147 MB of additional disk space will be used.
Get:1 http://archive.raspberrypi.org/debian stretch/main armhf libc-ares2 armhf 1.14.0-1~bpo9+1 [80.7 kB]
Get:2 http://mirror.rise.ph/raspbian/raspbian stretch/main armhf liblqr-1-0 armhf 0.4.2-2 [20.9 kB]
Get:3 http://archive.raspberrypi.org/debian stretch/main armhf libuv1 armhf 1.18.0-3~bpo9+1 [86.3 kB]
Get:4 http://mirror.rise.ph/raspbian/raspbian stretch/main armhf imagemagick-6-common all 8:6.9.7.4+dfsg-11+deb9u6 [184 kB]
Get:5 http://mirror.rise.ph/raspbian/raspbian stretch/main armhf libmagickcore-6.q16-3 armhf 8:6.9.7.4+dfsg-11+deb9u6 [1,571 kB]
Get:6 http://archive.raspberrypi.org/debian stretch/main armhf libwebsockets8 armhf 2.0.3-2+b1~rpt1 [85.2 kB]
[... 242 lines cut ...]
Processing triggers for hicolor-icon-theme (0.15-1) ...
Setting up libgs9:armhf (9.26a~dfsg-0+deb9u2) ...
Setting up libwmf0.2-7:armhf (0.2.8.4-10.6) ...
Setting up libboost-filesystem1.58.0:armhf (1.58.0+dfsg-5.1+rpi1+b1) ...
Setting up libc-ares2:armhf (1.14.0-1~bpo9+1) ...
Processing triggers for fontconfig (2.11.0-6.7) ...
Setting up libpcap0.8:armhf (1.8.1-3) ...
Setting up libwebsockets8:armhf (2.0.3-2+b1~rpt1) ...
Setting up mongodb-clients (1:2.4.14-4) ...
Setting up libmagickcore-6.q16-3:armhf (8:6.9.7.4+dfsg-11+deb9u6) ...
Setting up libmongoclient-dev:armhf (1.1.2-5) ...
Setting up mongodb-dev (2:1.1.2-5) ...
Setting up libjxr-tools (1.1-6) ...
Setting up libyaml-dev:armhf (0.1.7-2) ...
Setting up ghostscript (9.26a~dfsg-0+deb9u2) ...
Setting up mongodb-server (1:2.4.14-4) ...
Adding system user `mongodb' (UID 110) ...
Adding new user `mongodb' (UID 110) with group `nogroup' ...
Not creating home directory `/var/lib/mongodb'.
Adding group `mongodb' (GID 114) ...
Done.
Adding user `mongodb' to group `mongodb' ...
Adding user mongodb to group mongodb
Done.
```

bafu, 2019-10-27:
BerryNet keeps improving and fixing issues. If this issue still exists in the latest release, please share the steps to reproduce with me, and I will investigate it.

Please feel free to re-open this issue if it still can be reproduced.

closed 2019-10-27 by bafu