DT42/BerryNet #57: CNCF-Runtime discussion/presentation(?)

raravena80, 2020-10-02:
Hello BerryNet team,

I'm one of the co-chairs of the CNCF SIG-Runtime. I'm reaching out and I think it would be great for you to present/discuss the project at one of our meetings. For example, how can the project be used to run Cloud Native workloads? Can it be used with something like K3s?

Let me know if this something you'd be interested in doing. If yes, please feel free to add it to our [agenda](https://bit.ly/cncf-sig-runtime-meeting-notes) or reach out to me (raravena80 at gmail.com)

Thanks!

bafu, 2020-10-08:
@raravena80 Thanks for the invitation. I've replied an email, and see you online soon.

closed 2020-10-08 by bafu

raravena80, 2021-03-22:
Hi @bafu,

Checking in.  Would the team be interested in presenting at our [next](https://bit.ly/cncf-sig-runtime-meeting-notes) meeting?

Thanks!