DT42/BerryNet #59: [Bug Report]

tnguyen226-dxc, 2020-10-14:
I install Berrynet follow the manual installation. But after installed successfully the supervisor is not running camera and tflite
I checked the log files and see the error:
 pkg_resources.DistributionNotFound: The 'paho-mqtt' distribution was not found and is required by berrynet
I tried to install paho-mqtt by command: python3 -m pip install paho-mqtt
But still error,  camera and tflite are not running
Please advise how to fix this.
Thanks you

bafu, 2020-10-18:
@tnguyen226-dxc Thanks for the report, and I will try to reproduce this issue from my side. 

At the same time, there will be `berrynet-install.log` in your BerryNet directory. It can help reduce the debugging time if you don't mind to share it with me.

tnguyen226-dxc, 2020-10-18:
> @tnguyen226-dxc Thanks for the report, and I will try to reproduce this issue from my side.
> 
> At the same time, there will be `berrynet-install.log` in your BerryNet directory. It can help reduce the debugging time if you don't mind to share it with me.
This is the log file. Please help me check. Thanks you much 
[berrynet-install.log](https://github.com/DT42/BerryNet/files/5398346/berrynet-install.log)

bafu, 2020-10-20:
@tnguyen226-dxc I found this error message in `berrynet-install.log`:

`sudo: pip3: command not found`

You can try the steps below to fix this issue:

1. Install Python Pip

        $ sudo apt install python3-pip

1. Re-run the installation script

        $ ./configure

tnguyen226-dxc, 2020-10-20:
Hi ,
I followed your instruction and got the same error. This is new logs, please help me check.
Thanks you much.
[camera-stderr.log](https://github.com/DT42/BerryNet/files/5410934/camera-stderr.log)
[berrynet-install.log](https://github.com/DT42/BerryNet/files/5410936/berrynet-install.log)

bafu, 2020-11-01:
@tnguyen226-dxc You can resolve this issue by installing the `paho-mqtt` dependency manually:

    $ sudo apt install python3-paho-mqtt

or

    $ python3 -m pip install paho-mqtt

---
The dependency should be installed in the installation process, and I can not find the root cause from the logs yet. I will try to reproduce this issue from my end.

bafu, 2020-12-20:
This issue is fixed by PR #62.

**Root Cause**

Some Python Wheels did not be installed correctly only in Buster distribution.

In the Buster Docker file, these missing dependencies will be installed before executing the installation script (`configure`), so we did not notice this issue. Thanks for @tnguyen226-dxc 's report.

---
Verify commands:

```
pi@raspberrypi:~/codes/BerryNet $ bn_camera --debug --mode file --filepath ~/dog.jpg
[D 201220 04:08:48 camera:165] payload: 5.618 ms
[D 201220 04:08:48 camera:166] payload size: 234492
[D 201220 04:08:48 __init__:50] Send message to topic berrynet/data/rgbimage
[D 201220 04:08:48 camera:171] mqtt.publish: 12.582 ms
[D 201220 04:08:48 camera:172] publish at 2020-12-20T04:08:48.365974

pi@raspberrypi:~ $ DISPLAY=:0 bn_dashboard --no-decoration --no-full-screen --topic berrynet/engine/tflitedetector/result --debug
[D 201220 04:07:56 __init__:11] Connected with result code 0
[D 201220 04:07:56 __init__:13] Subscribe topic berrynet/engine/tflitedetector/result
[D 201220 04:08:49 __init__:20] Receive message from topic berrynet/engine/tflitedetector/result
[D 201220 04:08:49 fbdashboard:65] inference text result: {'timestamp': '2020-12-20T04:08:48.345930', 'annotations': [{'label': 'dog', 'confidence': 0.96484375, 'left': 128, 'top': 226, 'right': 320, 'bottom': 540}, {'label': 'bicycle', 'confidence': 0.953125, 'left': 110, 'top': 132, 'right': 568, 'bottom': 429}, {'label': 'car', 'confidence': 0.87890625, 'left': 459, 'top': 80, 'right': 694, 'bottom': 173}]}
```

closed 2020-12-20 by bafu