DT42/BerryNet #63: Can I run the darknet with multi-thread in the Raspberry？

huyutao3550346, 2021-01-05:
Hello everyone。
In the source code：https://github.com/DT42/BerryNet/blob/master/berrynet/service/darknet_service.py
darknet service and engine without multi-thread argue。
if I want run the darknet detect with mulit-thread like tflite，what should I do？

bafu, 2021-01-07:
@huyutao3550346 IIRC we enabled multi-thread in the low-level darknet library by default.

If you find that it's not true, please share the logs (e.g., `top`) with me and I will take a look.

You can also use [tfliteyolov4_service](https://github.com/DT42/BerryNet/blob/master/berrynet/service/tfliteyolov4_service.py) as an alternative.

huyutao3550346, 2021-01-07:
Thanks for your answer.
I add some code to print once running time.
if the darknet library enable the multi-thread by defult ,the time is too large. if cost 50s to run once nearly.
Is this normal? 
Is there any way to reduce time-consuming ?
There is the log. 
![new_code]
(https://user-images.githubusercontent.com/29395068/103876792-12668500-510f-11eb-9fdc-b3a51b31fefb.jpg)
[temp_darknet_log.txt](https://github.com/DT42/BerryNet/files/5780711/temp_darknet_log.txt)
I run it in the Raspberry 3B+ and without

closed 2021-07-07 by huyutao3550346